direc85/harbour-batterybuddy #8: Automatic chargin control does not suspend charging after limit has been reached

CosmicDuck, 2020-05-15:
SFOS 3.3.0.16
Sony Xperia X (Dual Sim)
BB 2.2-2

Automatic charging control is enabled
upper limit 85 %
lower limit 75 %

Use notification is enabled
Full notification 100 %
Low notification 25 %
Notification interval 1:00

BB is started and running, but charging will not suspend if the upper limit is reached and will charge up to 100 %.

CosmicDuck, 2020-05-15:
This is a bug

direc85, 2020-05-16:
Thanks for reporting. I experienced this once on my XA2 Ultra, too, but it has worked since...

Have you rebooted your phone since? What is the output of these commands (one should work):

    ls -l /sys/class/power_supply/battery/input_suspend
    ls -l /sys/class/power_supply/battery/charging_enabled
    ls -l /sys/class/power_supply/usb/charger_disable

After testing those; does stopping and resuming the charging manually work using the buttons?

And finally, does it work after reboot?

CosmicDuck, 2020-05-17:
```
[root@Sailfish nemo]# uptime
 20:00:07 up 1 day,  8:30,  1 user,  load average: 7.84, 5.90, 5.39
root@Sailfish nemo]# ls -l /sys/class/power_supply/battery/input_suspend
ls: /sys/class/power_supply/battery/input_suspend: No such file or directory
[root@Sailfish nemo]# ls -l /sys/class/power_supply/battery/charging_enabled
-rw-rw-rw-    1 root     root          4096 May 16 11:30 /sys/class/power_supply/battery/charging_enabled
[root@Sailfish nemo]# ls -l /sys/class/power_supply/usb/charger_disable
ls: /sys/class/power_supply/usb/charger_disable: No such file or directory
[root@Sailfish nemo]# ls -l /sys/class/power_supply/battery/
total 0
-rw-r--r--    1 root     root          4096 May 17 20:01 allow_hvdcp3
-rw-r--r--    1 root     root          4096 May 17 20:01 battery_charging_enabled
-rw-r--r--    1 root     root          4096 May 16 14:42 capacity
-r--r--r--    1 root     root          4096 May 17 20:01 charge_full
-r--r--r--    1 root     root          4096 May 17 20:01 charge_full_design
-r--r--r--    1 root     root          4096 May 17 20:01 charge_type
-rw-rw-rw-    1 root     root          4096 May 16 11:30 charging_enabled
-rw-r--r--    1 root     root          4096 May 17 20:01 constant_charge_current_max
-r--r--r--    1 root     root          4096 May 16 14:42 current_now
-r--r--r--    1 root     root          4096 May 17 20:01 cycle_count
lrwxrwxrwx    1 root     root             0 May 17 20:01 device -> ../../../02-qcom,qpnp-smbcharger
-rw-r--r--    1 root     root          4096 May 17 20:01 dp_dm
-rw-r--r--    1 root     root          4096 May 17 20:01 enable_shutdown_at_low_battery
-r--r--r--    1 root     root          4096 May 17 20:01 flash_active
-r--r--r--    1 root     root          4096 May 17 20:01 flash_current_max
-r--r--r--    1 root     root          4096 May 17 20:01 flash_trigger
-rw-r--r--    1 root     root          4096 May 17 20:01 fv_cfg
-rw-r--r--    1 root     root          4096 May 17 20:01 fv_cmp_cfg
-r--r--r--    1 root     root          4096 May 16 14:42 health
-r--r--r--    1 root     root          4096 May 17 20:01 input_current_limited
-r--r--r--    1 root     root          4096 May 17 20:01 input_current_max
-r--r--r--    1 root     root          4096 May 17 20:01 input_current_now
-r--r--r--    1 root     root          4096 May 17 20:01 input_current_settled
-rw-r--r--    1 root     root          4096 May 17 20:01 int_cld
-rw-r--r--    1 root     root          4096 May 17 20:01 lrc_enable
-rw-r--r--    1 root     root          4096 May 17 20:01 lrc_not_startup
-rw-r--r--    1 root     root          4096 May 17 20:01 lrc_socmax
-rw-r--r--    1 root     root          4096 May 17 20:01 lrc_socmin
drwxr-xr-x    2 root     root             0 May 17 20:01 power
-r--r--r--    1 root     root          4096 May 16 14:42 present
-rw-r--r--    1 root     root          4096 May 17 20:01 rerun_aicl
-rw-r--r--    1 root     root          4096 May 17 20:01 restricted_charging
-rw-r--r--    1 root     root          4096 May 17 20:01 safety_timer_enabled
-r--r--r--    1 root     root          4096 May 16 14:42 status
lrwxrwxrwx    1 root     root             0 May 17 20:01 subsystem -> ../../../../../class/power_supply
-rw-r--r--    1 root     root          4096 May 16 11:30 system_temp_level
-r--r--r--    1 root     root          4096 May 16 14:42 technology
-r--r--r--    1 root     root          4096 May 16 14:42 temp
-r--r--r--    1 root     root          4096 May 17 20:01 type
-rw-r--r--    1 root     root          4096 May 16 14:42 uevent
-rw-r--r--    1 root     root          4096 May 17 20:01 voltage_max
-r--r--r--    1 root     root          4096 May 17 20:01 voltage_max_design
-r--r--r--    1 root     root          4096 May 16 14:42 voltage_now

```
Maybe it's not compatible with latest OS or the Xperia X?

CosmicDuck, 2020-05-17:
```
[root@Sailfish nemo]#     ls -l /sys/class/power_supply/usb/               
total 0
-rw-r--r--    1 root     root          4096 May 17 20:06 current_max
lrwxrwxrwx    1 root     root             0 May 17 20:06 device -> ../../../78db000.usb
-rw-r--r--    1 root     root          4096 May 17 20:06 dp_dm
-rw-r--r--    1 root     root          4096 May 17 20:06 health
-rw-r--r--    1 root     root          4096 May 17 20:06 input_current_max
-rw-r--r--    1 root     root          4096 May 17 20:06 online
drwxr-xr-x    2 root     root             0 May 17 20:06 power
-rw-r--r--    1 root     root          4096 May 17 20:06 present
-r--r--r--    1 root     root          4096 May 17 20:06 scope
-r--r--r--    1 root     root          4096 May 17 20:06 sub_type
lrwxrwxrwx    1 root     root             0 May 17 20:06 subsystem -> ../../../../../class/power_supply
-r--r--r--    1 root     root          4096 May 17 20:06 type
-rw-r--r--    1 root     root          4096 May 16 14:47 uevent
-rw-r--r--    1 root     root          4096 May 17 20:06 usb_otg
-rw-r--r--    1 root     root          4096 May 17 20:06 voltage_max
-r--r--r--    1 root     root          4096 May 17 20:06 voltage_now
[root@Sailfish nemo]#     ls -l /sys/class/power_supply/usb/power/
total 0
-rw-r--r--    1 root     root          4096 May 17 20:07 autosuspend_delay_ms
-rw-r--r--    1 root     root          4096 May 17 20:07 control
-r--r--r--    1 root     root          4096 May 17 20:07 runtime_active_time
-r--r--r--    1 root     root          4096 May 17 20:07 runtime_status
-r--r--r--    1 root     root          4096 May 17 20:07 runtime_suspended_time

```

direc85, 2020-05-31:
> ```
> [root@Sailfish nemo]# ls -l /sys/class/power_supply/battery/charging_enabled
> -rw-rw-rw-    1 root     root          4096 May 16 11:30
> ```

Looks good - the file exists and permissions are correct!

> Maybe it's not compatible with latest OS or the Xperia X?

It is compatible with SFOS 3.3.0, I have used it myself. Someone had tested with Xperia X before IIRC, so it should be something small...

I just released Battery Buddy 3.0 with revised logic. Does the issue still exist?

CosmicDuck, 2020-06-03:
Problem solved after upgrading. Thx.

closed 2020-06-03 by CosmicDuck