direc85/harbour-batterybuddy #9: Notification sound should be splitted for upper and lower level notification

CosmicDuck, 2020-05-15:
I want notified only 1 time for upper limit but very often for lower limit.
Would be nice you can split them.
You can use the smallest interval to switch off the notification and longest interval to notifiy only one time. (- sign for never, ∞ for one time)

CosmicDuck, 2020-05-15:
This is a whish

direc85, 2020-05-16:
That's a nice suggestion, and makes perfect sense! It shouldn't be too difficult to implement, but it still takes some time to do it, as it requires many new configuration file entries and refined logic.

I use the application so that the alert limits are outside the charger limits, so I don't hear the alarm unless the battery is really empty, and never for the upper limit, because charging stops before that. It's good to receive other use cases - such as only using the alerts, or using the alerts inside the charging limits.

direc85, 2020-06-14:
Oops, I didn't actually implement the "once" behavior...

direc85, 2020-12-27:
Implemented in aea2ef3 and fine-tuned in 041adb3 and f582d55.

direc85, 2020-12-29:
Whoops! Should have been "Fixes issue 22" but this was fixed, too.

closed 2020-12-29 by direc85