direc85/harbour-batterybuddy #11: Add option to limit the usb current?

CosmicDuck, 2020-06-06:
Hello, maybe better to use a patch, but If usefull I want to a option which limits the max. usb current because I'm using a Qi-Pad with a limit of 1000 mA.
If this is not a good extension for the app, then it is also ok and I will check to add a patch with patchmanager.

direc85, 2020-06-06:
This is a nice idea and suits the application well - if the charging current is high enough, the battery heats up quickly, which shortens its life span.

The thing is; I don't know how to limit the current. I'll look into it soon.

direc85, 2020-06-14:
I don't think it's possible to limit the current itself. At least I can't find any way to do it.

The only option would be to to use "staircase" charging; charge for 1 minute, pause for 1 minute, repeat... The downside of that approach is that the device wakes up every time charging continues - I think - and so it wears down the device in other ways.

Perhaps it would be possible to disable waking up the device somehow, but that feels quite intrusive approach to me.

CosmicDuck, 2020-06-19:
I found a solution to limit the battery loading max current in µA:
[root@Sailfish nemo]# cat /sys/class/power_supply/battery/constant_charge_current_max 
1850000
Wait 1 minute
[root@Sailfish nemo]# cat /sys/class/power_supply/battery/current_now 
-779109

[root@Sailfish nemo]# echo -n 100000 >|/sys/class/power_supply/battery/constant_charge_current_max
Wait one minute

[root@Sailfish nemo]# cat /sys/class/power_supply/battery/current_now 
-809474
The change is minimal because my Qi-Pad is limited to 1 A and my
display was switched on with battery monitor running.

[root@Sailfish nemo]# echo -n 500000 >|/sys/class/power_supply/battery/constant_charge_current_max
[root@Sailfish nemo]# cat /sys/class/power_supply/battery/current_now 
-364988
Here I can see a relevant change.

current_now will only updated round about 1 time per minute. So mostly the change of limit is not seen.

For discussion: Maybe it is better to create a patch which extends the akku settings menu ;-)

CosmicDuck, 2020-06-19:
I found current_max in power/usb path. A lower limit looks like to solve my problem with the overload of the qi pad by using 
echo -n 1000000 >|/sys/class/power_supply/usb/current_max

direc85, 2020-06-28:
Thanks for researching this!

It's so frigging hot in Finland that I simply can't even function. I'll get into it when it gets cooler.

Fuzzillogic, 2020-11-18:
The app [Charger Control](https://openrepos.net/content/xkr47/charger-control) had the option to disable charging, but keep using USB as power supply for the phone. Ideally, if the battery reaches a certain charge threshold, say 75%, the charging should be stopped, but no battery would be used as long as the device has power from USB. When you disconnect, you're certain you have 75% charge left. I reckon no battery use at all results in even longer battery life, and the need for a lower charging limit can be removed.

See [source](https://github.com/xkr47/sailfish-charger-control/blob/master/qml/components/PowerModeController.qml). Unfortunately, this is only for specific older devices, like the original Jolla phone. I've messed around with the parameters @CosmicDuck mentioned on my XA2, but couldn't find a set of parameters equivalent to the wanted behavior. Any ideas if this is even possible with the current devices?

direc85, 2020-11-18:
Thanks for the additional info. I can look if I can find some way to control the charging, but I guess it's not possible. Not sure when I have the time to tinker with this, though...

rubdos, 2022-01-12:
Suggestion: planned charging. Add the capability to limit battery charging current during a programmable time slot, or to override "slow charging mode" with a tap in the Battery Buddy application when you actually *want* a quick charge. I almost never want quick charging on my phone :-)

direc85, 2022-12-31:
Fixed in 4.2.0 (and finalized in 4.2.1 :smiling_face_with_tear: )

closed 2022-12-31 by direc85

Fuzzillogic, 2023-01-01:
Thanks for the update. May I suggest to use this feature, by setting the charge current to 0mA, instead of suspending charging (`/sys/class/power_supply/battery/input_suspend`) when the upper charge threshold is reached? This way the battery won't "yo-yo" between charging and discharging when full, but is kept topped up. This might fix this issue I see when using my phone as navigation device, while hooked up to USB power: it charges up to the limit, but given the high load because of active navigation, it drains rather quickly to the lower limit, at which point it starts charging again. 

I just tried this when charging the battery: `# echo 0 > /sys/class/power_supply/battery/constant_charge_current_max` and then checked the current. The current was ~20mA (draining), so while not exactly 0, it drained much slower than when disconnected. And tweaking worked as well, e.g. setting a charge current of 50mA.

It could be combined with the current configured limits, and call it "slow charge mode within limits" or such. While I'm not _sure_ this is actually better for the battery longevity, to me it seems it puts less strain on the battery, and thus might be beneficial.

direc85, 2023-01-02:
Great idea! I turned your comment into an issue :)

rubdos, 2023-01-02:
That would turn Battery Buddy into Battery Lover ;D