direc85/harbour-batterybuddy #13: No feedback from Battery Buddy on Jolla 1

davodego, 2020-06-07:
I have a Jolla 1 and Battery Buddy is running but I dont notice any action from the app.
Automatic charging control is on. Use notification is on. Limits are set. All seems fine. But nothing happend. 
A status lookup gives:
'Active: failed (Resault: signal) since Sa 2020-06-06 16:23:32 CEST; 24h ago Main PID: 1003 (code=killed, signal=SEGV)

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.'

Any help or Ideas?

davodego, 2020-06-07:
``

direc85, 2020-06-08:
I think I know what's going on. And it just so happens that I have a Jolla Phone, so I should be able to recreate this one myself :)

direc85, 2020-06-14:
The reason - I believe - is the daemon trying to write to the charging file using null `QFile` pointer. This is fixed in commit 87e72be and released already in version 3.3 (the version bump commit was done way after the release).

Another issue was preventing the service(s) from starting after upgrade. I'm not really 100% totally sure what caused it, but at least on my phone, there were leftovers from the previous service, and cleaning those seems to have helped. This was fixed in commit 3e38f46 and also released in 3.3.

I also tested this with my Jolla phones (one with SFOS 3.3.0 and another with SFOS 2.2.1) and they both react as they should, and don't segfault.

Is the bug fixed for you with version 3.3?

davodego, 2020-06-14:
The problem is solved. I had a software bug that caused my Jolla to no longer start up. I had to do a factory reset the day before yesterday. After I imported all updates up to 3.3.0.16 again, I already installed version 3.3 without knowing it. The error no longer occurs. Batterybuddy is now doing what it should. Was it because of my Jolla or the update that it now works. I am thinking about the update because I had problems with Aliendalvik that led to the crash.

closed 2020-06-14 by davodego