direc85/harbour-batterybuddy #16: Charge/discharge to storage capacity and shutdown

martonmiklos, 2020-06-14:
Hello @direc85 

I happen to own a couple of "drawer" SFOS devices what I mainly keep in the drawer and pull out for testing/upgrading only.

As you may know the lithium batteries have a so called storage voltage which is the optimal for longer storage:
https://batteryuniversity.com/learn/article/how_to_store_batteries

I was thinking about creating a tool which would charge/discharge until the storage voltage is reached and the would shut down the device. By now that Battery buddy having a daemon mode this feature could be incorporated to it.

What do you think about this feature?

direc85, 2020-06-14:
It's a nice idea, but I don't think it works as-is... I can sure make the daemon issue a shutdown command, but the issue is that after the shutdown the phone is still connected to the charger, and will start up in the "charging mode". At that mode Battery Buddy is not active, and can't limit the charging, so the device will charge all the way to 100%...

martonmiklos, 2020-06-18:
That makes sense. But if we are below the storage voltage it might not be necessary to shutdown the device just limit the charge voltage to the storage voltage and push a notification when the storage charge is reached for e.g.

direc85, 2020-06-28:
That would essentially be charge limit by voltage, which is not necessarily an average user feature. I'll see how its implemented in some devices, and if it's feasible, I'll add it as an advanced feature.