direc85/harbour-batterybuddy #19: service not started on Sailfish os 3.4.0.24

ogustbiller, 2020-10-13:
Damon not started on Sailfish OS 3.4.0.24 , both buttons start and stop disable. There not process "batterybuddy-daemon" in process list.

hetas, 2020-10-18:
I'm on XA2 and I also have both buttons disabled but Buttery Buddy seems to work anyhow. I have the charge limiter on it's working.

I also don't have a process with that name running.

direc85, 2020-10-18:
Thanks for reporting.

I didn't notice that the service control buttons didn't work with SFOS 3.4, but for me both services work. The root service must work, because the charging pauses and resumes as set by the limits. It could be only the controls (perhaps running command line stuff from QML is broken?) are not working as expected.

I have been busy for the last few months, but it looks like I can spend more time with software development from now on.

direc85, 2020-10-19:
This has now been fixed in version 3.5.

For some reason it still requires either a restart of the device, or starting the service using the button control.

closed 2020-10-19 by direc85

ogustbiller, 2020-10-19:
> This has now been fixed in version 3.5.
> 
> For some reason it still requires either a restart of the device, or starting the service using the button control.

Thank you very much. Now work correct. 👍