direc85/harbour-batterybuddy #22: Add option to respect silence state of the phone

dryo, 2020-11-02:
Sometimes during the night I tend to sleep and wish to not be disturbed. To achieve this I set my phone to flight mode and put it to some silent ambience. This is shown by a small crossed out note in the top bar.

Unfortunately Battery Buddy does not respect this setting so it woke me up when the battery dropped below my low threshold.

At first I thought that Battery Buddy should always obey the "silent" setting but after thinking about it this should be made optional, at least for the battery low notification. Reason: When I use my phone as an alarm clock that I rely on and I forget to connect the phone to the charger overnight and the battery depletes, it's probably good when the phone wakes me up before the battery is empty. This would however be very annoying when being e.g. at a movie theater, so maybe make it optional.

What do you think?

direc85, 2020-11-02:
That is a good idea. I always put my phone in silent mode and keep it in charger overnight - alarm works still - and it didn't occur to me that the app isn't aware of silent mode... Nor ringtone volume, now that I think about it.

Will fix those next!

direc85, 2020-11-24:
As the volume of the notification should follow the set ringtone volume, it should then respect the silenced mode, too.

direc85, 2020-12-27:
This should be fixed in commit 07c463b but apparently `QSoundEffect::AudioRole` doesn't do the job as one could expect, and I have no idea what value I should write to `QMediaPlayer::Category` that was hinted in the mailing list...

direc85, 2020-12-27:
There is an upcoming hint called `sound-file` but it was [committed](https://git.sailfishos.org/mer-core/nemo-qml-plugin-notifications/commit/3ee9f656eb533ca53be074d533d6cc918818a82b) just 4 months ago, so perhaps in SailfishOS 3.5 bundling the sound in the notification finally fixes the issue for good...

direc85, 2020-12-29:
Fixed in commit 7b49027.

closed 2020-12-29 by direc85