direc85/harbour-batterybuddy #30: showing strange values in 4.0.1.48 (Koli)

bits-fritz, 2021-02-24:
Battery buddy is unusable with Sailfish OS 4.0.1.48 (Koli).
Attached screenshots show app values detailed and app with sailfish battery status icon.
Note: device is disconnected from charger.

Ready to answer questions for deeper investigation :o))

[image]
[image]

direc85, 2021-03-19:
I missed notifications from this issue completely, sorry about that!

I have no quick answer to what might be going on... I think this serves as the final kick for me to implement better debugging and data gathering tools inside the application - all optional, of course! I _think_ I have some free time tomorrow, I may be able to hack something up. No promises though...

Meanwhile, does Battery Buddy work for you otherwise (is it able to stop and resume the charging)? What device are you using? Is your username `nemo` or `defaultuser`? The latest version (at least...) is compatible with either username...

bits-fritz, 2021-03-20:
> I _think_ I have some free time tomorrow, I may be able to hack something up. No promises though...

sounds good! 
Good to see You're still alive instead of beeing hit by an electrical shock, while testing Battery Buddy.. ;-)

> Meanwhile, does Battery Buddy work for you otherwise (is it able to stop and resume the charging)?

Did it ever?

puhh! Looking at the facts I noticed I'm running version 1.3 installed from
Jolla store!!

Installing from open repos served the latest release 3.8
So the answer is: Yes load control works - very happy to see now this beeing implemented!

> What device are you using? 

Xperia10Plus-DualSIM

> Is your username `nemo` or `defaultuser`?

nemo

> The latest version (at least...) is compatible with either username...

That was the real Problem - upgrading solved everything - Yay!! :-)
Is there a way to update jolla store release or get a hint there?

You can close this one. 

Thank You for providing Battery Buddy!

direc85, 2021-03-20:
Haha, and there I was assuming you are running the latest version - didn't even ask for that :D

Great to hear it works now! I'm still going to develop better debugging amd reporting tools for the app, now that I _finally_ managed to get Sailfish Application SDK running SFOS 4.0.1.48 on all three architectures (TL;DR deleted all remnants of old installation and the configs, too, was required)

closed 2021-03-20 by direc85