direc85/harbour-batterybuddy #36: QTimer is inconsistent, when screen is locked

direc85, 2021-04-27:
The daemon uses QTimer to trigger re-reading the battery, charger et. al. files. This works like a clock (ha) when the device is in use. When the screen is locked - and the device in a low power state - the intervals can be delayed / skipped altogether. This needs to be fixed.

nephros, 2021-05-09:
This seems to be the API of choice in this case, according to https://forum.sailfishos.org/t/qml-timers-longer-than-10-minutes/

https://sailfishos.org/develop/docs/nemo-qml-plugins-keepalive/qml-nemo-keepalive-backgroundjob.html/

But, it's of course QML, not Qt.

direc85, 2021-05-22:
I found this thread over at TJC: https://together.jolla.com/question/175209/sudo-make-me-a-sandwich/

I'm currently experimenting with `keepalive-tool`, because that is the laziest possible solution. If it doesn't blow up battery consumption, I just might go with it... (Most likely it's not Jolla Store viable approach, but hey, I'm using a root service, so... :grin:)

direc85, 2021-07-25:
Well, the solution was studying https://git.sailfishos.org/mer-core/nemo-keepalive/ and switching from QTimer to BackgroundActivity.

Fixed in 5627c5f, released in 08b577f.

closed 2021-07-25 by direc85