direc85/harbour-batterybuddy #42: Implement battery temperature and health

direc85, 2021-05-02:
- [ ] Read battery temperature and health from file
- [ ] Dispay the data in GUI application
- [ ] Add relevant settings to control daemon behavior
- [ ] Implement notifications daemon

Verify functionality on different devices:

- [ ] Jolla Phone
- [ ] Xperia X
- [ ] Xperia XA2
- [ ] Xperia X10

direc85, 2021-05-02:
Work started in #40

direc85, 2021-05-02:
@nephros tested the file with Xperia X10, I tested it with Xperia XA2 (Ultra) - ticked two boxes more.

nephros, 2021-05-02:
> @nephros tested the file with Xperia X10, I tested it with Xperia XA2 (Ultra) - ticked two boxes more.

Nice! I develop on Xperia 10, so that should usually work. I can also test on Gemini, but as that doesn't have the charge control feature AFAIK that's of little use.

nephros, 2021-05-02:
possibility to check box number 3 in #43

nephros, 2021-05-09:
Just leaving a note here so I don't forget:

On Xperia10 there is /sys/class/power_supply/bms  (Battery Management/Monitoring System??), which has these:

    temp_cold
    temp_cool
    temp_hot
    temp_warm

Which seem to set the thresholds for the "health" reporting. Could be used (they are writable), but I'm not sure it's smart to expose them to users.

direc85, 2021-05-22:
It really is not a good idea to let users set the tresholds, but nice to know those exist :)

direc85, 2021-05-22:
The health and temperature files fortunately are in the same location across my devices (Jolla, X, XA2) and X10, too! Release gets closer and closer :)

The bad news is that `Notification.setSound` doesn't exist in SFOS 3.4 and earlier, so I'll have to create a separate branch for the older releases. I'm considering SFOS 3.0, because I just can't get the app sdk installing SFOS 2.2 build engine any more...

direc85, 2021-05-23:
Well, I ended up using defines and conditional compiling to reach that goal. I decided to use the legacy code path for SFOS 4 releases, too, since it still works fine (although I still feel it's a bit of a hack...)

Released in commit 1dd00da

closed 2021-05-23 by direc85