direc85/harbour-batterybuddy #64: Start/Changes do not work unless Start Background Service is enabled

sbridger, 2024-01-28:
Unless "Start Background Service" is enabled, "Start" does not.

Changing charge current slider, does not change actual charge current, unless the above.

So the bug is a silent fault: User changes slider - nothing happens. Pushes Start - nothing happens

X10III 4.5.0.24

direc85, 2025-03-16:
This is half-bug and half-feature. The service is masked by the switch, which doesn't stop it. It prevents start on reboot and manually starting the service, so it works fine in that sense. It's like that, because I couldn't get disabling and enabling the service work (something to do with permissions or sailjail perhaps?) but masking worked, so I went with that at the time.

I'll have a look at it. If nothing else, I'll make the start button disabled if the service is masked.

direc85, 2025-03-16:
Oh right, there was another bug as well.

Yes, currently the values changed have no effect, unless the service is running. This should be reflected as a warning label at the top of the settings page.